b-it-bots/mas_industrial_robotics #12: Simulation not working

abhishek098, 2018-03-12:
Changes made in Milan broke the simulation. I narrowed down the problem to commit #8 . Please revert changes.

sthoduka, 2018-03-12:
which commit?

abhishek098, 2018-03-12:
#8 this one.

abhishek098, 2018-03-12:
[image]

moriarty, 2018-07-13:
Warnings don't cause collisions... your ticket needs more information.

How did:
> Changes made in Milan broke the simulation.

Could you provide more description of:

1.  what you did
2. what you expected
3. what the actual result was

@abhishek098 #8 is a merge request, not a commit.
You should have
> narrowed down the problem

A little bit further than that.

The merge request appears okay... But I didn't test it, but from @sthoduka branch name: I assume this is the actual running state from ERL and was tested on the hardware:
So if it's the _running state on the hardware_ it **should not be reverted**. The hardware state is generally closer to the truth than the simulation.

You should submit a PR to fix the simulation.

The commit which would potentially cause that **Warning** is this one:
https://github.com/sthoduka/mas_industrial_robotics/commit/358d6b5371ef5cf834eeb70939ad028a82982196

But the MoveIt config files were regenerated and updated in this commit:
https://github.com/sthoduka/mas_industrial_robotics/commit/b79c838d4872d8f5ecf722171bcb881a4ec95bb5

I didn't test any of this... But the bringup sim reads the file from the first commit.

https://github.com/b-it-bots/mas_industrial_robotics/blob/929f457ddf37ba847e45e20a2f7be30d61294616/mir_simulation/mir_bringup_sim/robot.launch#L28

Or, maybe things are just out of date in your workspace, and that the new versions weren't installed.
Please ensure you've done a ```catkin clean``` and then ```catkin build``` which will install the newer xacro file into your development workspace.

Also: please provide the output of ```wstool``` (assuming you're using wstool) http://wiki.ros.org/wstool

The output should look something like this:

```

fetch@fetch3:~/melodic-stable/src$ wstool 
workspace: /home/fetch/melodic-stable/src

 Localname         S SCM Version (Spec)                           UID  (Spec)  URI  (Spec) [http(s)://...]
 ---------         - --- --------------                           -----------  ---------------------------
 robot_controllers   git indigo-devel  (-)                        06107134e86e github.com/fetchrobotics/robot_controllers.git
 robot_calibration   git master  (-)                              8c3c37895aa2 github.com/mikeferguson/robot_calibration
 power_msgs          git master  (-)                              78ec4d33f44c github.com/fetchrobotics/power_msgs.git
 opencv_candidate    git master  (-)                              749e709359e9 github.com/wg-perception/opencv_candidate.git
 navigation          git vrabaud-kinetic-tf2-math-on-melodic  (-) 88ada10d9129 github.com/ros-planning/navigation.git
 moveit_python       git master  (-)                              8dfbc9b2a30f github.com/mikeferguson/moveit_python.git
 joystick_drivers    git master  (-)                              493e5ad31c69 github.com/ros-drivers/joystick_drivers.git
 graft               git hydro-devel  (-)                         2dcee2016c4c github.com/ros-perception/graft.git
 fetch_ros           git indigo-devel  (-)                        3598ee057558 github.com/fetchrobotics/fetch_ros.git
 fetch_robots        git indigo-devel  (-)                        512be5394418 github.com/fetchrobotics/fetch_robots.git
 fetch_msgs          git master  (-)                              f4b10b9deff3 github.com/fetchrobotics/fetch_msgs.git
 fetch_gazebo      M git gazebo2  (-)                             d539bae7a911 github.com/fetchrobotics/fetch_gazebo
 fetch_drivers     M git master  (-)                              09db5da3d6ea amoriarty@sw44:~/melodic/src/fetch_drivers
 fetch_common        git master  (-)                              7bfb526f7f8b amoriarty@sw44:~/melodic/src/fetch_common

Also detected these repositories in the workspace, add using 'wstool scrape' or 'wstool set':

 Localname                SCM   URI  (Spec) [http(s)://...]
 ---------                ---   ---------------------------
 open_karto               --git git@github.com:ros-perception/open_karto
 slam_karto               --git git@github.com:ros-perception/slam_karto.git
 sparse_bundle_adjustment --git git@github.com:ros-perception/sparse_bundle_adjustment.git
```

So that @sthoduka can see if you have uncommitted changes **M** and which branches, and which commits you have checked out...

Otherwise @sthoduka or @deebuls should close the issue as invalid or cannot reproduce, because it's a few months old, and doesn't describe the problem in enough detail.

closed 2020-05-22 by deebuls