b-it-bots/mas_industrial_robotics #37: WBC segmentation fault

deebuls, 2018-06-13:
```

[ WARN] [/move_base]: The origin for the sensor at (2.02, 1.62) is out of map bounds. So, the costmap cannot raytrace for it.
[ INFO] [/mcr_perception/object_selector]: ======================================================
[ INFO] [/mcr_perception/object_selector]: Received request to select: CONTAINER_BOX_RED-01
[ INFO] [/mcr_perception/object_selector]: ******************************************
[ INFO] [/mcr_perception/object_selector]: Searching for: CONTAINER_BOX_RED-01 or CONTAINER_BOX_RED
[ INFO] [/mcr_perception/object_selector]: Current: CONTAINER_BOX_RED
[ INFO] [/mcr_perception/object_selector]: Found: CONTAINER_BOX_RED
[ INFO] [/mcr_perception/object_selector]: ******************************************
[ERROR] [/data_logger]: bad callback: <bound method DataLogger.pick_pose_cb of <mir_data_logger_ros.data_logger.DataLogger object at 0x7f89f70567d0>>
Traceback (most recent call last):
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/topics.py", line 750, in _invoke_callback
    cb(msg)
  File "/home/robocup/ros/robocup/src/unmerged_packages_for_testing/task_planning/mir_data_logger/ros/src/mir_data_logger_ros/data_logger.py", line 205, in pick_pose_cb
    msg.header.stamp = self.tf_listener.getLatestCommonTime("/map", msg.header.frame_id)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/tf/listener.py", line 101, in getLatestCommonTime
    return self._buffer.get_latest_common_time(strip_leading_slash(source_frame), strip_leading_slash(dest_frame))
InvalidArgumentException: Invalid argument passed to get_latest_common_time in tf2 frame_ids cannot be empty

[INFO] [/insert_object_server]: State machine transitioning 'GENERATE_OBJECT_POSE':'success'-->'CHECK_IF_OBJECT_IS_AVAILABLE'

tf2 frame_ids cannot be empty
[WARN] [/wbc]: Exception occurred: Invalid argument passed to get_latest_common_time in tf2 frame_ids cannot be empty
[WARN] [/wbc]: Exception occurred: Invalid argument passed to get_latest_common_time in tf2 frame_ids cannot be empty
Traceback (most recent call last):
  File "/home/robocup/ros/robocup/src/unmerged_packages_for_testing/manipulation/mir_whole_body_motion_calculator/ros/scripts/whole_body_motion_calculator_pipeline_class", line 6, in <module>
    mir_whole_body_motion_calculator_ros.whole_body_motion_calculator_pipeline_class.main()
  File "/home/robocup/ros/robocup/src/unmerged_packages_for_testing/manipulation/mir_whole_body_motion_calculator/ros/src/mir_whole_body_motion_calculator_ros/whole_body_motion_calculator_pipeline_class.py", line 355, in main
    whole_body_motion_calculator_pipeline.start()
  File "/home/robocup/ros/robocup/src/unmerged_packages_for_testing/manipulation/mir_whole_body_motion_calculator/ros/src/mir_whole_body_motion_calculator_ros/whole_body_motion_calculator_pipeline_class.py", line 138, in start
    state = self.idle_state()
  File "/home/robocup/ros/robocup/src/unmerged_packages_for_testing/manipulation/mir_whole_body_motion_calculator/ros/src/mir_whole_body_motion_calculator_ros/whole_body_motion_calculator_pipeline_class.py", line 186, in idle_state
    sampled_poses, base_motion_component = self.whole_body_motion_calculator.get_poses_and_base_motion(transformed_input_pose)
  File "/home/robocup/ros/robocup/src/unmerged_packages_for_testing/manipulation/mir_whole_body_motion_calculator/ros/src/mir_whole_body_motion_calculator_ros/simple_whole_body_motion_calculator_class.py", line 75, in get_poses_and_base_motion
    transformed_target_pose, base_motion_component = self.transform_target_pose(target_pose)
  File "/home/robocup/ros/robocup/src/unmerged_packages_for_testing/manipulation/mir_whole_body_motion_calculator/ros/src/mir_whole_body_motion_calculator_ros/simple_whole_body_motion_calculator_class.py", line 106, in transform_target_pose
    target_pose_frame = posemath.fromMsg(target_pose.pose)
AttributeError: 'NoneType' object has no attribute 'pose'
[WARN] [/pregrasp_planner_node]: Inbound TCP/IP connection failed: connection from sender terminated before handshake header received. 0 bytes were received. Please check sender for additional details.
[wbc-13] process has died [pid 9025, exit code 1, cmd /home/robocup/ros/robocup/src/unmerged_packages_for_testing/manipulation/mir_whole_body_motion_calculator/ros/scripts/whole_body_motion_calculator_pipeline_class ~dbc_pose_2:=/mcr_navigation/direct_base_cont

```

deebuls, 2018-06-21:
@mhwasil  did you solve this ? what commit was it ?

mhwasil, 2018-07-13:
@deebuls this was different problem while mine was in the object_list_merger which has been solved before Montreal. I didn't make any commit on this. Somebody did not publish the frame_id which is needed for tf.

deebuls, 2018-07-13:
@moriarty   thanks for the trace.
@mhwasil so can I close this issue ?

mhwasil, 2018-07-13:
@deebuls yes, please close it. We didn't have this issue in Montreal. Before going, someone was pushing all the changes on dbc.

closed 2018-09-30 by sthoduka