b-it-bots/mas_industrial_robotics #49: [melodic migration] mir_base_placement IGNORED, Depends on mcr_perception_msgs

abhishek098, 2018-11-27:
@mhwasil

mhwasil, 2018-11-27:
``Get:1 http://security.ubuntu.com/ubuntu bionic-security InRelease [83,2 kB]
Hit:2 http://de.archive.ubuntu.com/ubuntu bionic InRelease                     
Hit:3 http://de.archive.ubuntu.com/ubuntu bionic-updates InRelease             
Hit:4 http://de.archive.ubuntu.com/ubuntu bionic-backports InRelease           
Hit:5 http://packages.ros.org/ros/ubuntu bionic InRelease                      
Fetched 83,2 kB in 1s (127 kB/s)                  
Reading package lists... Done
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package ros-melodic-cob-scan-unifier
E: Unable to locate package ros-melodic-cob-srvs
E: Unable to locate package ros-melodic-dynamixel-controllers
E: Unable to locate package ros-melodic-dynamixel-msgs
E: Unable to locate package ros-melodic-eband-local-planner
E: Unable to locate package ros-melodic-gmapping
./repository.debs: line 64: rosinstall: command not found
./repository.debs: line 70: cd: ../youbot_driver: No such file or directory
./repository.debs: line 70: cd: ../youbot_driver_ros_interface: No such file or directory
./repository.debs: line 70: cd: ../youbot_description: No such file or directory
./repository.debs: line 70: cd: ../youbot_simulation: No such file or directory
./repository.debs: line 70: cd: ../youbot_diagnostics: No such file or directory
./repository.debs: line 70: cd: ../youbot-manipulation: No such file or directory
./repository.debs: line 70: cd: ../dynamixel_motor: No such file or directory
Reading package lists... Done
Building dependency tree       
Reading state information... Done
flex is already the newest version (2.6.4-6).
g++-multilib is already the newest version (4:7.3.0-3ubuntu2.1).
ros-melodic-tf2-geometry-msgs is already the newest version (0.6.5-0bionic.20181117.184336).
0 upgraded, 0 newly installed, 0 to remove and 234 not upgraded.
./repository.debs: line 22: rosinstall: command not found

sthoduka, 2021-05-20:
it seems like this has been fixed

closed 2021-05-20 by sthoduka