b-it-bots/mas_industrial_robotics #68: Not able to build the repository

santoshreddy254, 2019-08-18:


mhwasil, 2019-08-18:
mcr_algorithms package belongs to mas_common_robotics.
The issue has been addressed in [mas_common_robotics](https://github.com/b-it-bots/mas_common_robotics/issues/51)

santoshreddy254, 2019-08-25:
Errors     << mcr_cavity_detector:cmake /home/santosh/catkin_ws/logs/mcr_cavity_detector/build.cmake.003.log                                  
CMake Warning at /opt/ros/kinetic/share/catkin/cmake/catkinConfig.cmake:76 (find_package):
  Could not find a package configuration file provided by
  "mas_perception_msgs" with any of the following names:

    mas_perception_msgsConfig.cmake
    mas_perception_msgs-config.cmake

  Add the installation prefix of "mas_perception_msgs" to CMAKE_PREFIX_PATH
  or set "mas_perception_msgs_DIR" to a directory containing one of the above
  files.  If "mas_perception_msgs" provides a separate development package or
  SDK, be sure it has been installed.
Call Stack (most recent call first):
  CMakeLists.txt:4 (find_package)

Another error while building

santoshreddy254, 2019-08-26:
After pulling the unmerged packages from sydney-2019 branch
changed pcl_isnan() to std::isnan
There is a segment_scene wrong function call in pointcloud segmentation, extra argument – model_coefficients

Now errors are
Errors     << mir_object_recognition:make /home/santosh/b-it-bots-at-work/logs/mir_object_recognition/build.make.003.log                      
/home/santosh/b-it-bots-at-work/src/unmerged_packages_for_testing/perception/mir_object_recognition/ros/src/pointcloud_segmentation_ros.cpp: In member function ‘void PointcloudSegmentationROS::get3DBoundingBox(const ConstPtr&, const Vector3f&, mas_perception_libs::BoundingBox&, mas_perception_msgs::BoundingBox&)’:
/home/santosh/b-it-bots-at-work/src/unmerged_packages_for_testing/perception/mir_object_recognition/ros/src/pointcloud_segmentation_ros.cpp:184:46: error: invalid initialization of reference of type ‘mcr_perception_msgs::BoundingBox& {aka mcr_perception_msgs::BoundingBox_<std::allocator<void> >&}’ from expression of type ‘mas_perception_msgs::BoundingBox {aka mas_perception_msgs::BoundingBox_<std::allocator<void> >}’
     convertBoundingBox(bbox, bounding_box_msg);
@mhwasil @sthoduka

mhwasil, 2019-08-26:
can you move this issue to unmerged_packages_for_testing?
This package belongs to that repo

closed 2020-04-24 by deebuls