b-it-bots/mas_industrial_robotics #86: ROSPlan planning pipeline

DharminB, 2020-01-05:
Use [ROSPlan](https://github.com/KCL-Planning/ROSPlan)'s planning pipeline as much as possible instead of the custom developed one.

**Possible advantages**
- better support
- beginner friendly (because of ROSPlan's tutorial)
- removal of repeated code
- following "universal" standards

DharminB, 2020-01-06:
According to the conversation with @oscar-lima, `mir_pddl_problem_generator` should not be replaced with ROSPlan's pddl problem generator because it does not yet have a capability to select/favour a subset of goals which we need (3 goals at a time). 

The `RPActionInterface` cannot be used as a base classs because it interprets the effects of non durative actions as `at_start_effects` instead of `at_end_effects`. Thus a custom base action interface needs to be implemented. But that too wont be enough because there are specific functions like `rem_goals_at_location` in `mir_planner_executor` which needs to be integrated with the new implementation.

The planning and parsing using lama planner is not available in ROSPlan. This [pr](https://github.com/b-it-bots/ROSPlan/pull/3) takes care of it, which is a custom implementation as well. 

This in general sounds like a lot of re-implementation of the code which is already working and readable (more-or-less). Thus, I think, I will stop working/close on this [pr](https://github.com/b-it-bots/mas_industrial_robotics/pull/87) for now (for german open 2020) and focus on "more" pressing issues. However, someone else (who is interested in planning) is urged to work further on this. Here is a [link](https://github.com/DharminB/mas_industrial_robotics/tree/feature/rosplan_pipeline) to my fork's branch for future reference.

mhwasil, 2020-01-07:
can you update your PR and fix sphinx_rtd_theme since you already opened one. This is the fix #89