b-it-bots/mas_industrial_robotics #91: Continuous integration of mir docs into b-it-bots wiki

mhwasil, 2020-01-14:
Can we automatically push the built docs into [b-it-bots wiki](https://github.com/b-it-bots/wiki)? @Sushant-Chavan

Sushant-Chavan, 2020-01-16:
The built docs change their version number every time a mas_industrial_robotics get a new commit. This will cause a new commit to be pushed to the wiki repository where the difference is only in the version number as shown [here](https://github.com/Sushant-Chavan/wiki/commit/2fcce1c66d205945c1cfaf0ed63016a5718baf65). @mhwasil  how to avoid these version changes?

mhwasil, 2020-01-16:
The version of the docs is generated through [this function](https://github.com/b-it-bots/mas_industrial_robotics/blob/a7d77b2da55557fcc46d85fc641e99250f218f9f/docs/source/conf.py#L32)

Sushant-Chavan, 2020-01-16:
@mhwasil So can we hardcode this version for now?

mhwasil, 2020-01-16:
Yes, we will fix this once we have the release.

Sushant-Chavan, 2020-01-20:
Discussed the documentation versioning with @sthoduka and concluded that we will use the version name as `Kinetic` instead of the git commit hash id. This will avoid unnecessary commits in the `wiki` repository which could only be a change in the documentation version.

cc @mhwasil

PR #93 (2020-01-20): Build sphinx docs in CI
This PR will build the Sphinx documentation and push any updates to the `b-it-bots/wiki` repository.

## Changelog

* Hardcode the document version to `Kinetic` instead of git commit hash.
* Add a new stage to the travis build script to build the documentation

## Related PRs

Closes  #91 

## Checklist:

- [x] My code doesn't contain unnecessary comment blocks (e.g. unused code, templates of `package.xml` or `CMakeLists.txt`)
- [x] I have updated the `package.xml` and `CMakeLists.txt` with the correct dependencies.
- [x] I have updated the documentation accordingly.
closed 2020-01-20 by sthoduka