b-it-bots/mas_industrial_robotics #103: [manipulation] Calculate safe distance before perceiving drawer handle

DharminB, 2020-02-06:
Currently, the safe distance from the drawer is being hardcoded in `open_drawer_server` and `close_drawer_server` in package `mir_manipulate_drawer`.

The safe distance is the distance between front laser and drawer surface. The robot needs to make sure that it is far away from the drawer to move that arm without collision. The distance is hardcoded based on how far ahead is the arm gripper from laser when it is in the required orientation. This, however, can be calculated based on 
1) joint angles of the recorded arm orientation (here `drawer_handle_pre_grasp`)
2) transform from front laser to gripper

Currently only x value is needed but it would be better if all 3 values are calculated.

**Additional context**
https://github.com/b-it-bots/mas_industrial_robotics/pull/99#discussion_r375808251

sthoduka, 2020-03-09:
retrieve joint configuration for a named moveit config from the SRDF:
https://github.com/b-it-bots/mas_common_robotics/blob/kinetic/mcr_tools/mcr_moveit_commander_gui/ros/src/mcr_moveit_commander_gui/moveit_commander_widget.py#L138
I think the code there only returns the name, and not the joint configuration itself, but it should be possible to extend it to get the joint values too.

sthoduka, 2020-03-09:
Actually, ignore my comment above. You can use [srdfdom](https://github.com/ros-planning/srdfdom) to parse the SRDF (either from file or parameter server).