b-it-bots/mas_industrial_robotics #129: [simulation] Fix moveit

deebuls, 2020-04-24:
## Describe the bug

## Steps to Reproduce

1. ...
2. ...
3. ...
4. ...

## Expected Behavior

## Environment

* Version used (branch/commit):
* PC:
    - [ ] on the robot (if applicable specify which one)
    - [ ] developer's laptop
        * OS:
        * ROS version:
* build command:
    * [ ] catkin build
    * [ ] catkin_make

DharminB, 2020-04-27:
## Describe the bug
There are multiple errors when launching bringup_sim + moveit
- legacy mode namespace error in `roslaunch mir_bringup_sim robot.launch` 
```
  [ERROR] [1588002604.123247996, 0.001000000]: GazeboRosControlPlugin missing 
  <legacyModeNS> while using DefaultRobotHWSim, defaults to true.
  This setting assumes you have an old package with an old implementation of DefaultRobotHWSim, where the robotNamespace is disregarded and absolute paths are used instead.
  If you do not want to fix this issue in an old package just set <legacyModeNS> to true.
```
- gripper cannot be controlled in `roslaunch mir_moveit_youbot_brsu_1 move_group.launch` or `roslaunch mir_moveit_youbot_brsu_arm move_group.launch`
```
  [ WARN] [1588010764.306753167, 31.412000000]: Waiting for arm_1/gripper_controller// to come up
  [ WARN] [1588010770.342071184, 37.412000000]: Waiting for arm_1/gripper_controller// to come up
  [ERROR] [1588010776.375384485, 43.412000000]: Action client not connected: arm_1/gripper_controller//
```
- Fourth and fifth joint does not reach the target configuration in `rosrun moveit_commander 
moveit_commander_cmdline.py arm_1` > `go candle`
 Error in `move_group.launch`
```
  [ WARN] [1588011073.199744671, 333.948000000]: Controller arm_1/arm_controller failed with error code GOAL_TOLERANCE_VIOLATED
  [ WARN] [1588011073.199863838, 333.948000000]: Controller handle arm_1/arm_controller reports status ABORTED
```
[image]

However, after about 7-8 seconds joint 4 reaches its target but joint 5 does not (even after about a minute)
[image]

## Steps to Reproduce
1. `export ROBOT=youbot-brsu-1` in every terminal (or place it in a .bashrc or .zshrc file)
2. `roscore`
3. `roslaunch mir_bringup_sim robot.launch`
4. `rosrun moveit_commander moveit_commander_cmdline.py arm_1

## Expected Behavior
None of the errors should occur and the arm should reach the user's given target configuration and do so within reasonable time (2-5 seconds)

## Environment
* Version used (branch/commit): `kinetic`
* PC:
  
  * [ ]  on the robot (if applicable specify which one)
  * [x]  developer's laptop
    
    * OS: `linux (Ubuntu 16.04)`
    * ROS version: `kinetic`
* build command:
  
  * [x]  catkin build
  * [ ]  catkin_make

DharminB, 2020-04-27:
> legacy mode namespace error in `roslaunch mir_bringup_sim robot.launch`

Solved in https://github.com/mas-group/youbot_description/pull/9

> gripper cannot be controlled in `roslaunch mir_moveit_youbot_brsu_1 move_group.launch` or `roslaunch mir_moveit_youbot_brsu_arm move_group.launch`

Solved in #132

DharminB, 2020-04-27:
For third problem
> Fourth and fifth joint does not reach the target configuration in `rosrun moveit_commander moveit_commander_cmdline.py arm_1` > `go candle`

There is a [ROS answer](https://answers.ros.org/question/272325/moveitgazebo-controller-failed-with-error-code-goal_tolerance_violated/) suggesting that it might be PID gains problem. When I try `rosrun rqt_joint_trajectory_controller rqt_joint_trajectory_controller` as suggested in one of the comments in [the answer](https://answers.ros.org/question/272325/moveitgazebo-controller-failed-with-error-code-goal_tolerance_violated/?answer=272328#post-id-272328), joint 4 responds without any problem when the slider is moved but there is almost no movement for joint 5.

Could it be PID gains for our problem? @abhishek098

abhishek098, 2020-05-05:
Yes, the PID gains are the problem. I think @1apple7seeds had solved it if I remember correctly.

sthoduka, 2020-05-07:
The problem is actually the effort defined [here](https://github.com/mas-group/youbot_description/blob/kinetic-devel/urdf/youbot_arm/arm.urdf.xacro#L214). Changing it to `2.0` for example will fix the problem. 
I don't know if this will cause issue with the real robot though. My guess is it does not. See https://github.com/mas-group/youbot_description/commit/a9417b6cf76911229a5b79029f9e955aaa5a40df#diff-1df3b1bacbeec7006b364d83509abf46

sthoduka, 2020-05-07:
similarly changing the effort for joint 4 to `3.0` means it can go to `candle` with moveit

sthoduka, 2020-05-08:
After talking to @DjoleMNE I've reduced the friction and damping on the last two joints. The efforts should not be changed since it will be used if we use torque control.
https://github.com/mas-group/youbot_description/commit/4f4a8650bac6cd80e2c9677535831b9d55acf305

PR #132 (2020-05-09): Fixed problems with Moveit! simulation
## Changelog

- Added missing launch file for arm simulation without base
- Fixed gripper topic in yaml file for arm and youbot-brsu-1
- Removed launching of navigation and moveit (since it creates problems with launch of just arm or just base and also it creates inconsistency with `mir_bringup`)

## Related PRs

Closes #129  
Related to https://github.com/mas-group/youbot_description/pull/9 and https://github.com/mas-group/youbot_description/commit/4f4a8650bac6cd80e2c9677535831b9d55acf305

## Checklist:

- [x] My code doesn't contain unnecessary comment  [... cut]
closed 2020-05-09 by sthoduka