b-it-bots/mas_industrial_robotics #143: [Manipulation] Run pick state machine in simulation

deebuls, 2020-05-22:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

sthoduka, 2020-05-25:
Currently it is not possible to run the `pick_object_server` SM without having perceived the workstation, since the `object_selector` is triggered from the pick server. 
The options are:
1. Get `perceive_location` action running in simulation
2. Temporarily create a mockup `perceive_location` action, which will get the objects' positions from gazebo.

I haven't looked into the first option (which we should eventually do), but the second option is easy to do with the following services:
* `rosservice call /gazebo/get_world_properties "{}"` to get a list of models in the world
* `rosservice call /gazebo/get_model_state "model_name: '<model_name>'
relative_entity_name: 'youbot'"`to get the relative position of a model w.r.t the youbot. 
So we can get the objects which are "close" to the youbot and publish and object list with those.

DharminB, 2020-05-29:
[image]

We need to use a better realsense camera plugin for perception in sim

deebuls, 2020-07-03:
# Pending things:

[ ] MoveIT needs to be fixed
    - MoveIT doesnt say succceded