b-it-bots/mas_industrial_robotics #171: [perception] Tensorflow deprecated warnings

DharminB, 2021-03-12:
## Describe the bug

Upon launching planning bringup, tensorflow prints a large amount of warnings. It could be related to python2 vs python3 (but that is just a speculation).

## Steps to Reproduce

1. Start yb4 and make sure bringup runs without problems
2. Launch planning_bringup

* Version used (branch/commit):
* PC:
    - [x] on the robot (yb4)
    - [ ] developer's laptop
        * OS:
        * ROS version:
* build command:
    * [x] catkin build
    * [ ] catkin_make

DharminB, 2021-03-12:
```
{robocup@youbot-brsu-4-pc2}[ ~ ]
$ planning_bringup
... logging to /home/robocup/.ros/log/5acba60a-833b-11eb-8592-94c691a631a8/roslaunch-youbot-brsu-4-pc2-20863.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://youbot-brsu-4-pc2:34209/

SUMMARY
========

PARAMETERS
 * /amcl/base_frame_id: base_footprint
 * /amcl/first_map_only: True
 * /amcl/global_frame_id: map
 * /amcl/gui_publish_rate: -1
 * /amcl/kld_err: 0.01
 * /amcl/kld_z: 0.99
 * /amcl/laser_lambda_short: 0.1
 * /amcl/laser_likelihood_max_dist: 1.0
 * /amcl/laser_max_beams: 30
 * /amcl/laser_max_range: 5.0
 * /amcl/laser_min_range: 0.04
 * /amcl/laser_model_type: likelihood_field
 * /amcl/laser_sigma_hit: 0.2
 * /amcl/laser_z_hit: 0.95
 * /amcl/laser_z_max: 0.05
 * /amcl/laser_z_rand: 0.05
 * /amcl/laser_z_short: 0.01
 * /amcl/max_particles: 2000
 * /amcl/min_particles: 100
 * /amcl/odom_alpha1: 7.0
 * /amcl/odom_alpha2: 5.0
 * /amcl/odom_alpha3: 5.0
 * /amcl/odom_alpha4: 5.0
 * /amcl/odom_alpha5: 7.0
 * /amcl/odom_frame_id: /odom
 * /amcl/odom_model_type: omni
 * /amcl/recovery_alpha_fast: 0.0
 * /amcl/recovery_alpha_slow: 0.0
 * /amcl/resample_interval: 2
 * /amcl/transform_tolerance: 0.1
 * /amcl/update_min_a: 0.01
 * /amcl/update_min_d: 0.05
 * /amcl/use_map_topic: True
 * /arm_moveit_client/arm: arm_1
 * /arm_moveit_client/end_effector_frame: gripper_static_gr...
 * /arm_moveit_client/goal_joint_tolerance: 0.005
 * /arm_moveit_client/goal_orientation_tolerance: 0.01
 * /arm_moveit_client/goal_position_tolerance: 0.01
 * /arm_moveit_client/loop_rate: 10.0
 * /arm_moveit_client/move_group: move_group
 * /gripper_controller/grasp_monitor/load_threshold: 0.16
 * /gripper_controller/grasp_monitor/position_error_threshold: 0.07
 * /gripper_moveit_client/arm: arm_1_gripper
 * /gripper_moveit_client/goal_joint_tolerance: 0.0001
 * /gripper_moveit_client/goal_orientation_tolerance: 0.0001
 * /gripper_moveit_client/goal_position_tolerance: 0.0001
 * /gripper_moveit_client/loop_rate: 10.0
 * /gripper_moveit_client/move_group: move_group
 * /mcr_common/dynamic_reconfigure_client/config_file: /home/robocup/mel...
 * /mcr_common/dynamic_reconfigure_client/loop_rate: 10.0
 * /mcr_navigation/direct_base_controller/coordinator/base_frame: base_link
 * /mcr_navigation/direct_base_controller/coordinator/idle_loop_rate: 1.0
 * /mcr_navigation/direct_base_controller/coordinator/loop_rate: 20.0
 * /mcr_perception/cavity_finder/target_frame: base_link_static
 * /mcr_perception/cavity_pose_selector/DISTANCE_TUBE: R20
 * /mcr_perception/cavity_pose_selector/F20_20_B: F20_20
[... 965 lines cut ...]
[mir_perception/back_camera/pointcloud_to_laserscan-21] killing on exit
[mir_perception/back_camera/barrier_tape_detection-20] killing on exit
[mir_perception/front_camera/pointcloud_to_laserscan-19] killing on exit
[mir_perception/front_camera/barrier_tape_detection-18] killing on exit
[mcr_common/dynamic_reconfigure_client-17] killing on exit
[mcr_navigation/direct_base_controller/coordinator-16] killing on exit
[waypoint_trajectory_generation-15] killing on exit
[pregrasp_planner_node-14] killing on exit
[wbc-13] killing on exit
[static_transform_publisher_node-12] killing on exit
[gripper_controller/grasp_monitor-11] killing on exit
[move_base_wrapper-10] killing on exit
[move_base-9] killing on exit
[amcl-8] killing on exit
[rear_laser_scanner_repub-7] killing on exit
[front_laser_scanner_repub-6] killing on exit
[map_server-5] killing on exit
[pose_visualiser-4] killing on exit
[gripper_moveit_client-3] killing on exit
[arm_moveit_client-2] killing on exit
[move_group-1] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete

```

mhwasil, 2021-03-12:
@DharminB some codes are deprecated in the version we use.