b-it-bots/mas_industrial_robotics #248: Issue building python_orocos_kdl dependency repo

MannavaVivek, 2022-07-26:
## Describe the bug

Building the entire workspace for b-it-bots will sometimes throw an error when building the python_orocos_kdl package. The error message as been attached to this issue. 
[image]

## Steps to Reproduce

1. It is a system specific issue and couldn't be successfully replicated.

## Expected Behavior

The package should build without issues.

## Environment

* Version used (branch/commit):
* PC:
    - [ ] on the robot (if applicable specify which one)
    - [ ] developer's laptop
        * OS: Ubuntu 20.04
        * ROS version: ROS1 Noetic 

* build command:
    * [x] catkin build
    * [ ] catkin_make

## Temporary solution
Once all the repositories of the dependencies have been downloaded to the system, checkout the commit ID 6480921b236baacf282d32857c9a4a1c6f9bd971 of the orocos_kinematics_dynamics repo. 

Refer to the issue opened on the original repo for more details [Issue #401](https://github.com/orocos/orocos_kinematics_dynamics/issues/401)

deebuls, 2022-11-13:
Isnt this when you build the whole package you get this error. And then we you compile again the error is gone.

MannavaVivek, 2022-11-13:
No, it happened with the then latest commit of the repo. It happened regardless of how many time you build. And as far as I could see from the attached issue page, this issue still exists.

MannavaVivek, 2023-02-23:
Can confirm that the issue is now resolved. Latest builds of the workspace from scratch doesn't throw this error.

kvnptl, 2023-05-21:
While building the repo from scratch, got `fatal error: pybind11/pybind11.h: No such file or directory` error

MannavaVivek, 2023-05-21:
I recently did a new install and it did not occur for me. If this is occuring at random, then it would mean the dependency on the pybind version. We need to identify a version that works and add it to the requirements file.

closed 2023-10-26 by sthoduka