b-it-bots/mas_industrial_robotics #270: Design a staging area for yb2 configuration

kvnptl, 2023-02-15:
- design should be capable of holding 3 objects

GokulKrishna02, 2023-02-17:
Initial workflow design:

[image]

RavisankarSelvaraju, 2023-06-08:
initial test design with preliminary tests proves sufficient for the staging area. 
Todo:
- Reduce material in the staging area design 
- Find a material/board to fix the staging areas to the platform 
- adjust the staging area design as suggested by @KishanSawant

RavisankarSelvaraju, 2023-06-08:
[image]
[image]
[image]

kvnptl, 2023-06-15:
TODO: update a real robot picture with a staging area here

GokulKrishna02, 2023-06-17:
- Final Staging area fixed on YouBot2.
[image]

closed 2023-06-17 by GokulKrishna02