b-it-bots/mas_industrial_robotics #271: Grasp verification during unstaging

kvnptl, 2023-02-15:


RavisankarSelvaraju, 2023-03-24:
[image]

[image]

RavisankarSelvaraju, 2023-06-24:
The grasp verification is implemented during unstaging and it will retries 3 times. but other failure recovery stratesies are not implemented yet.

kvnptl, 2023-06-25:
So if the robot picks up and stages a ghost object, then the robot would try to unstage it 3 times and then move to the next task. Is it correct?

kvnptl, 2023-06-25:
https://github.com/b-it-bots/mas_industrial_robotics/issues/297

kvnptl, 2023-06-25:
#297

kvnptl, 2023-06-25:
related #297

RavisankarSelvaraju, 2023-06-25:
> So if the robot picks up and stages a ghost object, then the robot would try to unstage it 3 times and then move to the next task. Is it correct?

This is the observed behaviour during one  of the recent runs. I haven't looked into the code yet. I can update on this tomorrow