b-it-bots/mas_industrial_robotics #281: Infinite warning messages while running 2dnav

kvnptl, 2023-02-17:
After running `roslaunch mir_2dnav 2dnav.launch` command, it keeps on printing warning messages on the terminal.

## Describe the bug

Because of the error message, it's difficult to read other information on the terminal.

## Steps to Reproduce

1. Run `roslaunch mir_bringup robot.launch`
2. Run `roslaunch mir_2dnav 2dnav.launch`

## Expected Behavior

Sample warning message:
```
at line 278 in /tmp/binarydeb/ros-noetic-tf2-0.7.5/src/buffer_core.cpp
Warning: TF_REPEATED_DATA ignoring data with redundant timestamp for frame base_footprint at time 1676652917.751665 according to authority unknown_publisher
```

## Environment

* Version used (branch/commit): noetic
* PC:
    - [x] on the yb2 robot
        * OS: Ubuntu 20.04
        * ROS version: Noetic
* build command:
    * [x] catkin build

deebuls, 2023-03-16:
There was a issue someone had found that this is a problem in some library which cannot be solved. ? Do you have the link for that issue @kvnptl ?

kvnptl, 2023-03-16:
https://github.com/ros/geometry2/issues/467#issuecomment-1465756970 

still, some active discussion going on this issue