b-it-bots/mas_industrial_robotics #293: [Dataset collection] new Dataset for RoboCup2023

kvnptl, 2023-05-19:
- 7 basic object set (existing)
- 9 advanced object set (new)

Total: 16 objects

kvnptl, 2023-05-19:
**Basic object set**

[image]

kvnptl, 2023-05-19:
**Advanced object set** (9 objects)

[image]

[image]

kvnptl, 2023-05-22:
- Please find the sheets for dataset collection (set1 and set2) [HERE, BIB-CLOUD](https://bib-cloud.bib.hochschule-bonn-rhein-sieg.de/apps/files/?dir=/Shared/b-it-bots-ds/atwork/images/object_detection/YOLO/robocup_2023_dataset/dataset_collection&fileid=16051529)
- Dataset collection id  allotment [HERE](https://bib-cloud.bib.hochschule-bonn-rhein-sieg.de/apps/files/?dir=/Shared/b-it-bots-ds/atwork/images/object_detection/YOLO/robocup_2023_dataset/dataset_collection&openfile=16051640)

Let me know, if you have any doubts

GokulKrishna02, 2023-05-27:
- 0 - 154 dataset collection from set 1 completed.
- Dataset collected on different surfaces and on RTT and shelf.

kvnptl, 2023-05-29:
**Guidelines for data collection:**
1. Keep the camera on tripod
    - Maintain the same height and angle as Youbot's front camera 
3. Use `rosrun image_view image_view image:=<image topic>` to see camera feed
   - Use right-click to save images (NOTE: saved image names start from frame0001)
   - Rename the images according the id(s) given in the sheet (set1 or set2)
4. Keep the lights ON in the room
5. Use different surfaces
    - [image]
6. Use DECOY objects with our at-work objects, as shown in the image
    - [image]

7. Regarding where and how to place objects on the workstation
    - **Where**: keep objects in the 20 cm range on the workstation [check the below image]
    - **How**: keep at least 2 cm gap between the objects [check the below image]
    - [image]

kvnptl, 2023-05-31:
- IDs 308 to 461, dataset collection from set 1 and set 2 completed and updated [HERE](https://bib-cloud.bib.hochschule-bonn-rhein-sieg.de/apps/files/?dir=/Shared/b-it-bots-ds/atwork/images/object_detection/YOLO/robocup_2023_dataset/dataset_collection/KEVIN_RAVI&fileid=16214720).

kvnptl, 2023-05-31:
- Use [file_renaming.py](https://bib-cloud.bib.hochschule-bonn-rhein-sieg.de/apps/files/?dir=/Shared/b-it-bots-ds/atwork/images/object_detection/YOLO/robocup_2023_dataset/dataset_collection&openfile=16217151) script for renaming files

kvnptl, 2023-06-04:
[UPDATE]

The initial YOLOv8(s) model trained on image ids 308-461 (both sets) is ready to use for further annotation. Please find the trained model [HERE](https://bib-cloud.bib.hochschule-bonn-rhein-sieg.de/apps/files/?dir=/Shared/b-it-bots-ds/atwork/images/object_detection/YOLO/robocup_2023_dataset/dataset_collection/01_Trained_Models&fileid=16267283).

In order to use the custom trained model in `anylabeling` tool,
1. Download the ONNX and YAML files from [HERE](https://bib-cloud.bib.hochschule-bonn-rhein-sieg.de/apps/files/?dir=/Shared/b-it-bots-ds/atwork/images/object_detection/YOLO/robocup_2023_dataset/dataset_collection/01_Trained_Models&fileid=16267283)
    - NOTE: keep both files in the same directory
2. In `anylabeling` tool, click on "Auto Labeling" icon -> from the drop-down menu, select "Load Custom Model" -> select the downloaded YAML file.

Please let me know if you face any issues.

GokulKrishna02, 2023-06-07:
- 0 - 154 set 2 annotations completed and updated [HERE](https://bib-cloud.bib.hochschule-bonn-rhein-sieg.de/apps/files/?dir=/Shared/b-it-bots-ds/atwork/images/object_detection/YOLO/robocup_2023_dataset/dataset_collection/GOKUL_VIVEK/set2_000_154_annotated&fileid=16260155).

kvnptl, 2023-06-07:
[FINAL UPDATE]

- Dataset collection completed
- Dataset annotation completed

=> **Find the full dataset [HERE](https://bib-cloud.bib.hochschule-bonn-rhein-sieg.de/apps/files/?dir=/Shared/b-it-bots-ds/atwork/images/object_detection/YOLO/robocup_2023_dataset&fileid=16051420)**

closed 2023-06-07 by kvnptl