b-it-bots/mas_industrial_robotics #296: Push all the 3D designs and STL files to a common repo/bibcloud

RavisankarSelvaraju, 2023-06-08:


kvnptl, 2023-06-08:
Uploaded Derby cycle adaptor designs

kvnptl, 2023-06-15:
@GokulKrishna02 add camera mounts design on the bib cloud

GokulKrishna02, 2023-06-17:
Uploaded back camera mount designs to bibcloud.

kvnptl, 2023-06-17:
Pending design to upload:
- staging area
- gripper controller mount
- gripper mount
- gripper finger mount
- gripper design (flexible material design)
- battery base mount

kvnptl, 2023-06-20:
Gripper controller mount

<p align="left">
  [image]
  <br></br>
</p>

RavisankarSelvaraju, 2023-06-21:
- Staging area design 
- Gripper finger mount 
- gripper design 

these 3 design are updated in bib cloud

GokulKrishna02, 2023-06-23:
Uploaded below designs to bibcloud:
- battery base mount
- gripper controller mount 
- gripper mount

closed 2023-06-23 by GokulKrishna02