b-it-bots/mas_industrial_robotics #306: Implement New Battery Management System

GokulKrishna02, 2023-09-19:
- We need to implement a new battery management system in YouBot to use hot-swap of battery.

harleylara, 2023-09-30:
To be able to hot-swap batteries we need to define some parameters, assuming 50 W of power consumption under high workload for the NUC (nominal 30W, [reference](https://www.intel.com/content/www/us/en/products/sku/121637/intel-nuc-12-pro-kit-nuc12wski7/specifications.html)):

- `max hot-swap time window`: how long should the system remain on? with this parameter we can determine the capacity needed for the backup battery to keep the NUC and Gripper powered up.

## Connection scenarios

Parallel connection between the main battery and backup battery
- Pros:
  - Easy to implement, it is just matter of connecting both batteries in parallel.
- Cons:
  - Voltage balancing time, it take time for both batteries to reach the same voltage.
  - The backup battery will never charge completely due the discharging of the main battery -unless both are charging but in this case the robot is not operational-, if the backup battery is not getting full charging cycles the life time of the battery gets reduced.

Both (main an backup) are completely independent
- Pros:
  - No balancing time.
  - Independent charging cycles to maximize battery life time.
- Cons:
  - More difficult to implement. In this case we need a "watchdog" monitoring under voltage conditions that switch between main and backup.
  -  It is now necessary to monitor and charge two batteries, main and backup.

## Alternative solution 

Alternative solution for "hot-swapping". Based on the youBot manual section 2.1 "[...] power plug (labeled as 24V IN) can be used to power the KUKA youBot and recharge the battery (at the same time)." what if at the time to swap the main battery we just plug in the power supply intro the port `24V IN` (to keep the system on) and then replace the battery and finally remove the power cable?

1. Plug in the power supply to `24V IN`
2. Swap batteries
3. Unplug the power supply
4. Done

harleylara, 2023-10-02:
After a conversation with @vamsikalagaturu :

## Rationale

The main idea of a new battery manage system is to be able to carry batteries on international flights, due to airline limitations it is not possible to travel with the current battery (Derby Cycle Li-on 25.2V 325Wh), so the idea is to use multiple batteries equivalent to the current capacity.

## Airlines conditions

> You are allowed to carry portable electronic devices (such as watches, cameras, phones, laptops…) that
contain lithium metal or ion cells or batteries for your personal use under the following conditions:

- They should be carried in your carry-on luggage, although they may be in your checked baggage if needed as long as you take measures to prevent unintentional activation.
- The battery must not exceed a Watt-hour (Wh) rating of 100 Wh or 2 grams of lithium content (the first limit is for rechargeable lithium-ion batteries and the second for lithium metal batteries, which are usually not rechargeable). To calculate Watt-hours, just multiply the battery voltage by the Amp hours (Ah), as the Wh rating is not marked on them.
- If the Wh is higher than 100 but not higher than 160, you will need an approval from the operator to carry the item. It is not allowed to transport any item which battery exceeds 160 Wh.

List of of conditions EASA: [More practical information – Lithium batteries](https://www.easa.europa.eu/sites/default/files/dfu/Background%20Information%20Lithium%20Batteries_update%20Sept%202017.pdf)

For batteries exceeding 100 Wh but max. 160 Wh
- Maximum of two batteries permitted in carry-on baggage.

For batteries less than 100 Wh:
- It depends on the airline but it **is allowed to carry more than TWO** of this capacity. For example Airfrance allows up to 20 batteries less then 100 Wh.

Airlines batteries information (as reference):
- [Lufthansa information about batteries](https://www.lufthansa.com/content/dam/lh/documents/prepare-for-your-trip/baggage/dangerous-goods/Batteries_restrictions.pdf)
- [Airfrance prohibited items](https://wwws.airfrance.de/en/information/bagages/produits-interdits-et-reglementes)

## Regulations

- [Lithium Batteries in Baggage - FAA USA](https://www.faa.gov/newsroom/lithium-batteries-baggage)
- [Dangerous Goods - EASA EU](https://www.easa.europa.eu/en/domains/passengers/dangerous-goods)
- [Lithium Batteries - EASA EU](https://www.easa.europa.eu/community/topics/lithium-batteries-0)

harleylara, 2023-10-02:
# Battery Requirements 

- Ability to be transported on international flights (check this [coment](https://github.com/b-it-bots/mas_industrial_robotics/issues/306#issuecomment-1742766288))
- Comply with current battery characteristics
- Nominal autonomy time: 40 min

## Current Battery Information

- Battery type: Li-ion
- Nominal voltage: 25.2 V
- Power: 625 Wh
- Weight: 3 Kg

harleylara, 2023-10-02:
# Power Considerations

Arm:
- Voltage: 24V DC
- Drive train power limitable to : 80 W (value from [wiki](http://www.youbot-store.com/wiki/index.php/YouBot_Detailed_Specifications#Arm))
- Max limitable current: 3.34 A
- 
NUC:
- Voltage: 12-19V DC
- Max. Power: 90W

Gripper:
- Voltage: 12.6V DC
- Max. Power: 30W

## Measurements

Idle (mobile base not moving)
- Voltage: 24.4 V
- Current: 3A

Mobile base running (max speed):
- Voltage: 24.4V
- Current: 3.8A

Base and arm moving:
- Voltage: 24.4V
- Current: 4.2A
- ℹ️ Nominal Power: 100W

harleylara, 2024-04-23:
After a fruitful experience at the RoboCup German Open 2024 in which we won not only the first place but also gained experience, we were able to collaborate and talk with Christoph Steup from robOTTO and ask them about their cool hot swap battery system here are some insights:

- They use two **18V 4Ah Li-ion Bosch batteries for power tools**
- A simple diode for protection of **_reverse current_** between batteries
- Two DC-DC converters (unspecified power)
  - Step up to 24V
  - Step down to 12V

## Hardware Options

- Battery: [Bosch GBA 18V 5Ah - Obi Market](https://www.obi.de/p/4707949/bosch-professional-akkupack-gba-18-v-5-ah)
- Charger: [Bosch GAL 18 V-40 - Obi Market](https://www.obi.de/p/7940448/bosch-professional-akku-ladegeraet-gal-18-v-40-solo)
- DC-DC converter: for these we must check the power distribution and consumption to select the right hardware.

@Akhilan-xd if you can pin me in Slack or WhatsApp we can check about this.

Akhilan-xd, 2024-04-23:
@harleylara Sure, Harley we can start working on this. I will also search about the battery type and costs. So, that we can further proceed on the battery hot swapping for youbot.

harleylara, 2024-04-24:
# Current Power Distribution

The following diagram represents the power distribution and the different voltage levels used in the robot.

[image]

# Proposed Power Distribution

The proposed distribution system allows battery replacement without rebooting the system, and even allows switching to an external power supply (for testing in lab) without rebooting.

[image]

## Proposed Hardware

New components:
- Battery: [Bosch GBA 18V 5Ah - Obi Market](https://www.obi.de/p/4707949/bosch-professional-akkupack-gba-18-v-5-ah)
- Charger: [Bosch GAL 18 V-40 - Obi Market](https://www.obi.de/p/7940448/bosch-professional-akku-ladegeraet-gal-18-v-40-solo)
- DC-DC converter: [RSD-300D-24 DC/DC converter, 300W, 24V, 12.5A](https://www.reichelt.de/de/en/dc-dc-converter-300w-24v-12-5a-rsd-300d-24-p256809.html)
- Ideal diode: TBD

## Some Notes

In order to reduce costs with the DC-DC converter the RSD-300D-24 was proposed, the main disadvantage is that it is relatively bulky **40x96.5x216 mm** but the footprint fits inside the robot. In case we want to reduce weight or space you could opt for more compact alternatives such as the **TracoPower TEP 200WIR** , the main problem is that _**the price increases by a factor of 3**_.